guidosch/bikecounter #13: Floating interrupt pin (PIR-sensor)

MeierTobias, 2021-10-30:
Does the bug still exist after the improvements?
If yes: Add an external pull-down resistor (ex. 10kOhm)

guidosch, 2021-10-31:
May be we could also try to power the PIR-sensor over a GPIO and powercycle if we encouter the problem again.

MeierTobias, 2021-10-31:
> 
> 
> May be we could also try to power the PIR-sensor over a GPIO and powercycle if we encouter the problem again.

I've allready implemented this functionality in commit 56dffbbf8f9618892b4d479a88fb53bb3e4537cf
You can now specify the GPIO pin to power the sensor. (currently pin 2)
I also found an internal pull-down resistor flag. I thought that the MKR 1310 only has internal pull-up resistors. But its worth a try :)

This issue was meant as a "follow up" but I think we can close it and reopen a new one if the problem still exists.

closed 2021-10-31 by MeierTobias

guidosch, 2021-10-31:
First device is resoldered and uses the new code with the power pin. Works so far nicely.

MeierTobias, 2021-10-31:
Wuhuu 🚲