guidosch/bikecounter #21: Determen measurement accuracy

MeierTobias, 2021-10-30:


guidosch, 2022-01-06:
Etwas ist mir gestern noch aufgefallen. Wir hatten ja im Serial Log einige mehrfach Zählungen vom PIR Sensor. Könnte es sein, dass im Retrigger Mode der PIR Sensor zu lange auf "High" ist und dann die CPU mehrfach aufweckt? Oder braucht es immer einen Wechsel von "Low" auf "High" damit ein Count gezählt wird?

MeierTobias, 2022-01-06:
Es braucht immer einen wechsel von low auf high (rising edge) damit der Interrupt ausgeführt wird. Ich habe aber auch schon festgestellt, dass beim Senden der Daten der PIR-Sensor ausschlägt. Zuerst dachte ich es sei einfach Rauschen auf dem interrupt Pin aber die Messung mit dem Oszilloskop hat gezeigt das der Sensor in diesem Moment eine saubere Flanke asugibt. 

Ich habe auch einmal bemerkt, dass die 3.3V Output Spannung beim Senden zusammen bricht. (Manchmal bis auf 2.7V herunter)
Habe dies versucht mit einem Kondensator zu überbrücken hat aber leider nichts genützt.
Da wir die Spannungsversorgung des PIR-Sensors jetzt ja über einen DO-Pin machen, könnten wir den Sensor vor dem Senden ausschalten und danach wieder ein. Das könnte das Problem lösen.
Ich werde das im neuen Prototypen einmal testen.

MeierTobias, 2022-01-06:
Für die Bestimmung der Messunsicherheit könnten wir neben den Testfahrten auch drei oder vier Zähler an der selben Stelle montieren und die Abweichungen untereinander vergelichen.