guidosch/bikecounter #24: Implement Lora downlink

MeierTobias, 2021-10-30:
- ensures that the message was transmitted
- pass configuration to device

guidosch, 2022-01-06:
Idea is to use the downlink to check if the message was received by any gateway. If the device does not receive a "ACK" it will retry a few times (not endless du to battery drain).

May collect failed messages as counter and add to health data to get a idea how many times it does not work. Could also be determined by the msg. send intervall...

MeierTobias, 2022-03-09:
modem.endPacket(true);

MeierTobias, 2022-04-13:
I tested and implemented the modem.endPacket(true); method into the pro version.
It works well but I saw some uplink messages which were sent in a very short time span (20 sec.) I want to check if this has something to do with the "ACK" downlink

MeierTobias, 2022-09-08:
implemented and tested with #48

closed 2022-09-08 by MeierTobias