guidosch/bikecounter #48: Method to use internal RTC

MeierTobias, 2022-04-09:
With time sync over downlink message

guidosch, 2022-04-14:
We use the low power lib to make a deep sleep when only listening to the GPIO event from the PIR. Does the internal RTC work in deep sleep?

MeierTobias, 2022-04-14:
I had a look at the documentation and the deepSleep mode should not affect the RTC but I have not tested it yet. It's a bit hard to check because the deepSleep messes with the serial connection. So the best way to test this is by sending the timestamp to the ttn and check it there.

MeierTobias, 2022-04-14:
I'll test this on my breadboard after I finished the "allmendtrail-1" counter

MeierTobias, 2022-04-15:
approach: (just a note for myself)

- device sends its local time within the data package
- uplink payload formatter calculates the deviation between the device and the server time and passes the package to the webhook
- the cloud function gets the data package and checks if the deviation is in the acceptable range (+ stores the data)
- if the deviation exceeds the threshold then a downlink message will be enqueued with the needed adjustment parameters

MeierTobias, 2022-08-29:
To do: (note for myself)

- [x] remove external rtc code
- [ ] implement internal rtc
- [ ] add timestamp to data package
- [ ] process timestamp on server and enqueue downlink message
- [ ] extract and apply the deviation information from the downlink message
- [ ] Change the time handler from DateTime to (int)seconds and remove all dependencies of the RTClib.h

MeierTobias, 2022-09-08:
all finished and tested

closed 2022-09-08 by MeierTobias