guidosch/bikecounter #51: Timestamp for health data

guidosch, 2022-04-14:
At the moment the timestamp for the health data is taken from the first counter hit in the timestamp array.

This makes it easy to code the loop and every database entry can be used to sum up couter hits.

On the oder hand the timestamps on the health graph look strange, as the timestamps are not in a periodical order. Can we leave this or should we change it?
https://console.cloud.google.com/functions/details/europe-west6/storeBikeCounterPro?env=gen1&project=bikecounter

MeierTobias, 2022-04-14:
But for the second issue I metiond (count=0) I could change the device code so that I sends the current time stamp instead of an empty time array

MeierTobias, 2022-04-15:
I thought again about the synchronization problem #48 and I think to solve this in an elegant way the device should send its current time stamp within every data package. This time stamp could then also be used to display the health data

guidosch, 2022-04-17:
Great idea and already implemented.

closed 2022-04-17 by guidosch