guidosch/bikecounter #79: Round battery voltage to calm down graph

guidosch, 2023-01-22:
The voltage level toggels between some micro levels which makes the graph unreadable.

[image]

MeierTobias, 2023-02-26:
I would suggest to keep as much information as possible in the database for future debugging issues. (Also a voltage step discretization of less than 0.1V is necessary to determine the battery level.)

One possible solution would be to adjust the graph scales to a fix range. 
- Bat. volt. 3V - 4.2V
- Bat. level 0% - 100%) 

I think this would tidy up the graph enough to make it better readable.

guidosch, 2023-02-27:
Scaltes to fixed range did the trick.

closed 2023-02-27 by guidosch