guidosch/bikecounter #80: Entries in DB with date 1970-01-01

guidosch, 2023-02-17:
There are many entries in the DB with timestamp=0. Couting showed something like 30 entires only for antennentrail-1.
We may should also send the timestamp from TTN to have an idea when this happens.

[image]

guidosch, 2023-02-17:
Dates are not 0 but 1970-01-20, hmmm even more strange.

guidosch, 2023-02-18:
Not only Antennentrail but also all the others with latest software version.

MeierTobias, 2023-02-26:
I think this problem is already solved. 
The first package of the device is the time synchronization message. This contains a wrong timestamp (uninitialized) which triggers the cloud function to send a downlink message with the correction parameter. I updated the cloud function to check if the received message is a sync message and if so to not save it into the DB. But I think this change was applied a few weeks after the software update due to the lack of permissions to edit the cloud functions. 
I think these values are artifacts from that period where the cloud function did not handle that special case.

So I would say it is save to delete these entries.

MeierTobias, 2023-02-26:
Related fix #62

guidosch, 2023-02-26:
Deleted timestamps with year 1970. Closing issue.

closed 2023-02-26 by guidosch