guidosch/bikecounter #81: Store and visualize all receiving gateways.

guidosch, 2023-02-27:
May helps to detect issues with periods without data.

guidosch, 2023-06-06:
Data from Adlisberg is received by the following gateways:

http://www.openstreetmap.org/?mlat=47.3727869&mlon=8.46114635&zoom=12 (uetikon)
http://www.openstreetmap.org/?mlat=47.4427157809935&mlon=8.32905651030692&zoom=12 (Spreitenbach)
http://www.openstreetmap.org/?mlat=47.374&mlon=8.548&zoom=12 (Uni Zürich Zentrum)
http://www.openstreetmap.org/?mlat=47.382&mlon=8.541&zoom=12 (Zürich Letten)
http://www.openstreetmap.org/?mlat=47.41975053708852&mlon=8.368210494518282&zoom=12 (Spreitenbach)
http://www.openstreetmap.org/?mlat=47.3453013928798&mlon=8.57385246027386&zoom=12 (Zollikon)
http://www.openstreetmap.org/?mlat=47.359&mlon=8.56&zoom=12 (Zürich Mühlebach)
http://www.openstreetmap.org/?mlat=47.349428040199165&mlon=8.584124785753692&zoom=12 (Zollikerberg)

guidosch, 2023-06-06:
`const fetch = require('node-fetch');

const gateways = ['uzhloragwkol1',
    'uzhloragwbot',
    'uzhloragwstb'];

//loop over all gateways
gateways.forEach(gateway => {
    fetch("https://eu1.cloud.thethings.network/api/v3/gateways/" + gateway + "?field_mask=name,description,antennas,location_public", {

        "headers": {
            "accept": "application/json, text/plain, */*",
            "accept-language": "de-CH,de;q=0.9,en-GB;q=0.8,en;q=0.7,de-DE;q=0.6",
            "authorization": "Bearer yourToken",
            "sec-ch-ua": "\"Google Chrome\";v=\"113\", \"Chromium\";v=\"113\", \"Not-A.Brand\";v=\"24\"",
            "sec-ch-ua-mobile": "?0",
            "sec-ch-ua-platform": "\"macOS\"",
            "sec-fetch-dest": "empty",
            "sec-fetch-mode": "cors",
            "sec-fetch-site": "same-origin",
            "cookie": "_session=QABijk7; _csrf=MTY4NjA3ADpriFmXKhg_I",
            "Referer": "https://eu1.cloud.thethings.network/console/gateways/zueritrails-lora-gw/location",
            "Referrer-Policy": "strict-origin-when-cross-origin"
        },
        "body": null,
        "method": "GET"
    }).then(res => res.json())
        .then(json => {
            const id = json.ids.gateway_id;
            const name = json.name;
            const description = json.description;
            let latitude  = 0;
            let longitude = 0;
            let altitude = 0;
            if (json.antennas != null && json.antennas.length > 0) {
                latitude = json.antennas[0].location.latitude;
                longitude = json.antennas[0].location.longitude;
                altitude = json.antennas[0].location.altitude;
            }
            console.log(id + "," + name + "," + description + "," + latitude + "," + longitude + "," + altitude);
            //http://www.openstreetmap.org/?mlat=latitude&mlon=longitude&zoom=12
            const url = `http://www.openstreetmap.org/?mlat=${latitude}&mlon=${longitude}&zoom=12`
            console.log(url);
            console.log("\n");

        });
});

// payload: {"ids":{"gateway_id":"uzhloragwbot","eui":"B827EBFFFE4E1CE7"},"created_at":"2021-12-09T09:27:45.114Z","updated_at":"2021-12-09T09:28:53.695Z","name":"uzhloragwbot","description":"UZH-ZI-ZW, uzhloragwbot","antennas":[{"location":{"latitude":47.359,"longitude":8.56,"altitude":453,"source":"SOURCE_REGISTRY"},"placement":"OUTDOOR"}],"location_public":true}

`

guidosch, 2023-06-06:
/Users/guidoschnider/programming/firebase-direct/queryGateways.js

guidosch, 2023-06-06:
`const fetch = require('node-fetch');

const gateways = ['uzhloragwkol1',
    'uzhloragwbot',
    'werke-dlos8-001',
    'ncazrh-gw-1',
    'eui-a84041ffff21f300',
    'hb9eue',
    'b827ebfffe97f686',
    'fcc23dfffe0f1787',
    'eui-b827ebfffe29dbb1',
    'uzhloragwstb'];

//loop over all gateways
gateways.forEach(gateway => {
    fetch("https://eu1.cloud.thethings.network/api/v3/gateways/" + gateway + "?field_mask=name,description,antennas,location_public", {

        "headers": {
            "accept": "application/json, text/plain, */*",
            "accept-language": "de-CH,de;q=0.9,en-GB;q=0.8,en;q=0.7,de-DE;q=0.6",
            "authorization": "Bearer yourToken",
            "sec-ch-ua": "\"Google Chrome\";v=\"113\", \"Chromium\";v=\"113\", \"Not-A.Brand\";v=\"24\"",
            "sec-ch-ua-mobile": "?0",
            "sec-ch-ua-platform": "\"macOS\"",
            "sec-fetch-dest": "empty",
            "sec-fetch-mode": "cors",
            "sec-fetch-site": "same-origin",
            "cookie": "_session=MTY4NTkwOTI4OHw4MUqQABijk7; _csrf=MTY4NjA3NjY2NAgOhiHgoLzrJyI8daADpriFmXKhg_I",
            "Referer": "https://eu1.cloud.thethings.network/console/gateways/zueritrails-lora-gw/location",
            "Referrer-Policy": "strict-origin-when-cross-origin"
        },
        "body": null,
        "method": "GET"
    }).then(res => res.json())
        .then(json => {
            const id = json.ids.gateway_id;
            const name = json.name;
            const description = json.description;
            let latitude  = 0;
            let longitude = 0;
            let altitude = 0;
            if (json.antennas != null && json.antennas.length > 0) {
                latitude = json.antennas[0].location.latitude;
                longitude = json.antennas[0].location.longitude;
                altitude = json.antennas[0].location.altitude;
            }
            console.log(id + "," + name + "," + description + "," + latitude + "," + longitude + "," + altitude);
            //http://www.openstreetmap.org/?mlat=latitude&mlon=longitude&zoom=12
            const url = `http://www.openstreetmap.org/?mlat=${latitude}&mlon=${longitude}&zoom=12`
            console.log(url);
            console.log("\n");

        });
});

//payload {"ids":{"gateway_id":"uzhloragwbot","eui":"B827EBFFFE4E1CE7"},"created_at":"2021-12-09T09:27:45.114Z","updated_at":"2021-12-09T09:28:53.695Z","name":"uzhloragwbot","description":"UZH-ZI-ZW, uzhloragwbot","antennas":[{"location":{"latitude":47.359,"longitude":8.56,"altitude":453,"source":"SOURCE_REGISTRY"},"placement":"OUTDOOR"}],"location_public":true}

`

MeierTobias, 2023-12-09:
Done withing #98

closed 2023-12-09 by MeierTobias