guidosch/bikecounter #93: Improve LoRa connection logic

MeierTobias, 2023-07-31:
Move the whole connection logic to its own class/state machine and correct the waiting and idle behavior.

If this does not help change to the https://github.com/mcci-catena/arduino-lorawan library which is a high level abstraction of the https://github.com/mcci-catena/arduino-lmic which provides an adapted LoRaWAN-MAC-in-C library (from IBM)

MeierTobias, 2023-08-26:
Antenna test results: 
[Arduino Antenna](https://store.arduino.cc/products/dipole-pentaband-waterproof-antenna):
In the attic at my house (vertical alignment)
|                    |  T1 |  T2 |  T3 | T4 |  T5 | T6 |  T7 |  T8  | T9 | T10 | T11 | T12 | avg |
| ------------ | - |  - |  - |  - |  - |  - |  - |  - | - | - |  - |  - | - |
| RSSI [dBm] |  -119 | -105 | -118 | -105 | -105 | -108 | -103 | -107 | -118 | -118 | -111 | -107 | **-107.5** |
| SNR[dBm] |  -6.5   | -10.25 | -9.25 | -10.5 | -10.25 | -8.5 | -10.75 | -10 | -9.5 | -6.5  | -10.5 | -9 | **-9.04** |

guidosch, 2023-09-08:
Below a screen shot of the July on the Adlisberg Trail with fresh loaded batteries. It first was online a few days then offline and then felt again in the endless sending mode and the offline again. Later no more data but the batteries where drained within just a few weeks by doing something...
[image]