guidosch/bikecounter #99: Missing "L" in hoeckler-new-3 collection

MeierTobias, 2023-11-28:
The collection name of the Höckler Trail is hoecker-new-3 instead of hoeckler-new-3.
Are there any dependencies to this name on the UI side?

I did not notice that before and wrote hoeckler-new-3 into the connection table. That's the reason why the device was reported as offline today. I changed it to  hoecker-new-3.

MeierTobias, 2023-11-28:
I did not find an easy solution for moving the created documents from the collection hoeckler-new-3 into the original hoecker-new-3 collection.

Here is a script that should perform the required steps.
[https://stackoverflow.com/a/47263246](hoeckler-new-3)
But maybe you have a better solution/idea.

guidosch, 2023-11-28:
I have a script which I used a few times to move data to an other named data base. As we have the mapping table now I could do this. Before I would have to update the device first and then do the move.

guidosch, 2023-12-10:
Adjusted UI code and moved all data to the correct collection "hoeckler-new-3 "

guidosch, 2023-12-10:
Done

closed 2023-12-10 by guidosch